RymansatSatelliteTracker/RST #6: FM衛星の送受信周波数の設定異常

jh4xsy, 2025-03-24:
FM衛星でIC-9700のリグコンを試したら、送受信周波数が予期した値より大きくなった。

Satellite Mode: Normal, Auto: Onで、IC-9700がサテライトモードになり、送受信周波数はSO-124の既定値：145.925MHz/436.885MHzに設定されたが、送信周波数は AOS前:145.925MHz⇛MEL:145.618MHz⇛LOS:145.831MHz と変化した。

IC-9700との接続は、Device: IC-9700, Serial Port: /dev/ttyUSB0, Borate: 115200 baud と設定して、「Test Connect」ボタンにて"Serial connection successful"を確認済み。

RSTはLinux環境で作成したAppImageを使用。

jh4xsy, 2025-03-24:
2025/3/24日10:32の設定周波数は下図の通りでした。Ground Stationの緯度経度はQM06cbから設定してます。

[image]

kei-Shima, 2025-03-24:
ご連絡ありがとうございます。周波数設定がおかしい問題は複数報告をいただきましたので修正させていただきます。対応に少しお時間をいただければと思います。

jh4xsy, 2025-03-26:
ドップラーシフト補正値(Rx)を、useTransceiverCtrl.tsにて表示設定すると、Gpredictが表示するDopplerに近い数値が得られ、桁違いの補正をしてないことが確認できた。

```
09:52:23.610 › [main] ドップラーシフト補正値(Tx): -0.003kHz
09:52:23.613 › [main] ドップラーシフト補正値(Rx): 0.008kHz
```
[image]

画面表示を良く観察すると、受信周波数がドップラーシフト補正値の分だけ変化するようだ（その時点の受信周波数にドップラーシフト補正値を足しているように見える）。だとすると、MELになっても規定ビーコン周波数の値に一致しない結果(既報)にも符合します。

jh4xsy, 2025-04-21:
issue 11対応のマージ版でカイゼンがありました。2025/4/21日の結果は以下の通り。このパスでは、リグのダイアルは操作してません。
[image]

jh4xsy, 2025-04-30:
v0.1.2がリリースされたので、本件を再確認してみた。

今回は、受信周波数の表示値をRSTとGpredictとで比較した（IC-9700の表示値まで記録できない）。測定はSO-124 4/30日11時のパスで、このときIC-9700のダイヤルには触ってない。

その結果、表示周波数の差異は高々40Hzで、一事例一般化になるけど、充分な計算精度だと考えます。

|time|RST(MHz)|Gpredict(MHz)|Δf(Hz)|
|---|---|---|---|
|11:09:00|436.894941|436.894935|6|
|11:11:00|436.893140|436.893101|39|
|11:13:00|436.883137|436.883117|19|
|11:15:00|436.876070|436.876047|23|
|11:16:59|436.874925|436.874922|3|

IC-9700の周波数更新が追いつかない問題は、別issueを起こします。

closed 2025-05-01 by jh4xsy