RymansatSatelliteTracker/RST #7: 初めて使ってみました。

JF1PTU, 2025-03-24:
先ずはマニュアルを読まないで直感的にやってみました。（Ver0.1を評価）普段はSatPC32を使っています。

無線機設定：
　IC-9700を接続。通信速度はUSBケーブル接続なので115.2kbpsを選択し接続テストボタン押下で接続成功の表示。
　ただし、あえて異なるCOMポート番号にして接続テストするが、接続成功の表示の後、数秒後にピンク色のエラー表示が
　でた。接続成功の表示の表示条件は？エラー表示は正しい動作ですが。

ローテータ設定：
　Fox Delta社のST-2（USB）を接続。通信速度は規定の9600bpsを選択し接続テストボタン押下で接続成功の表示。
　無線機と同様に異なるCOMポート番号でテストするも同じ現象で接続成功表示の後、エラー表示。

　テストモードでNを0度でAz（方位）の←左ボタン押下でマイナス表示となるがST-2はマイナス表示データは無視するので
　動かない。手動で359度を書き込み、←左ボタン押下では動作する。-1度は359度と変換して動作させる必要がある。
　ローテータコントローラはKENPRO製のS（180度）⇔N（360度/0度）⇔S（180度）
　Az(方位）の左ボタン、右ボタンを押し続けた場合、間欠動作で動くのでぎこちない。
　一方、El（仰角）の下ボタンではマイナス表示はするがST-2は無視して動かない。マイナス表示は必要か？　0でストップが
　正し
　いか？なぜか、上または下ボタンを押し続けてた場合、間欠ではなく連続でスムースに動く。

　パーク（待機位置）設定が欲しい。現状は設定なしで、Az:北（0度）、El:90度（天頂）となっているが、Elは0度（水平）が
　望ましい。また現状は、LOS後、即パーク設定が動作している。何分前から自動で動かす設定があるがLOS後も、その設定と
　同じ時間で良いのでEl＝0でAz（LOS後の値）の位置で維持後、パーク設定位置に戻して欲しい、

　現状は、ノーマルモード固定になっているようであるが、フリップモードが欲しい。できれば自動でノーマルかフリップかを
　選択できるようにして欲しい。

衛星関連設定：
　表示衛星とTLE読み込みの関係が解らない。TLEで読み込みでアマチュア衛星関係のURLのみ選択しているが、初期設定？の
　衛星がすべて表示されている。不要な衛星は削除したいが？　あるいは選択したURLの衛星のみ表示して欲しい。

　Groupで選択した衛星の一つをダブルクリックでマニュアル設定でその衛星の周波数情報を書き込めるが、一組のみ書き込み
　二組くみ目を書き込まないでOKをしてもインバリットエラーになる時がある。また、ダウンリンク周波数のみの場合（ビーコ
　ンなど）はアップリンク周波数を書き込まないとエラーになる。0を書き込んでもエラーとなる。また右端の〇欄で周波数を
　選択できるようになっているがアップリンクとダウンリンクはペアになっていなく別々に選択できてしまうのでまずい。
　周波数設定は二つのモード以上の設定もあり（クロスバンド、ビーコン、SSTVなど）要検討。別Groupを作って対応？
　DATAモード（USB-D、LSB-D、FM-D）のモード選択が欲しい。MIC選択、USB端子データ選択があるので。

実際の運用：
　AO73、ISSで周波数制御、ローテータ制御をやってみた。ローテータ制御、レーダチャート表示はうまく動作しているようで
　あるが、周波数制御がおかしく、正しい周波数にならない。CAT制御がうまくいっていないようである。
　CATのフォーマットデータ（パケット的）の送信待ちの遅延設定がいるのでは？
　衛星固定（送受ともドップラー制御）、受信固定（送信のみドップラー制御）、送信固定（受信のみドップラ制御）の
　ドップラーモード設定がまだない。

　　　　　　　　　　　　　　　　　　　　　　　　　　　　　　　　　　　de　JF1PTU　大澤

kei-Shima, 2025-03-25:
ご連絡ありがとうございます。シリアル接続は今のところ接続ができたかどうかを見ていて、接続さえできれば成功となります。その後無線機に対してコマンドを送りますがタイムアウトしているのでエラーが出る状況です。
接続時に無線機かどうかの確認を入れる検討をしたいと思います。

ローテーターのAz手動で0度からのマイナスへはコントローラによって動きが違いそうですが動作を想定して359度にするかどうか検討させていただきます。

モードの追加はいくつまであればよいですかね？　SDR Consoleだといくつでも追加できそうですが。現在はどんどん追加できるようなUIになっていないのでたくさんあるようであればUI変更も検討したいと思います。

周波数はアップとダウンでペアでなければいけないのはトランスポンダ以外の1周波数しかない衛星の場合という解釈でよいでしょうか？
トランスポンダは周波数が違うので別々でよいかと思います。ただし、トランスポンダのほうが少ないので通常は1周波数なのに、設定が分かれていてペアでないのはユーザーの誤解を招くということでしょうか？

DATAモード（USB-D、LSB-D、FM-D）はマニュアルを見てどう実装するか検討させていただきます。

以下の要望は以前からありましたので順次対応させていただきます。
- 　フリップモード、自動ノーマル・フリップ選択モード
- 　衛星固定（送受ともドップラー制御）、受信固定（送信のみドップラー制御）、送信固定（受信のみドップラ制御）

JF1PTU, 2025-03-25:
接続確認（ACK応答の有無で判断？）とコマンドが通るかどうかの2段階は致し方なしですかね。

周波数表記については、クロスバンド通信（FMは周波数固定、SSB/CWは帯域中心周波数）、送受同一バンド通信（APRS等、
ただしドップラ周波数制御がプラスマイナスの関係なので送受別表示）なのでペアの関係になります。〇のクリックは連動必要。
（現状は独立事象の選択になっている？）。ビーコン、デジトーカ、SSTVなどは受信周波数のみで送信周波数は無し。
SatPC32では一行単位でのダウンリンク/アップリンク周波数（無しの場合はゼロ表記）、ダウンリンクモード/アップリンクモード（無しの場合は同じモード表記）、NORまたはREV（アップリンク無しの場合はNOR）、ダウンリンクオフセット/アップリンクオフセット、説明コメントをTXTベースで書いています。これを枠表示で1行にできるか？
ISSでは7行の記述がありました。通常は三つあればと思います。

クロスバンド通信はサテライトモード(送受同時通信）ですが同一バンド通信（APRSなど）は通常の無線機のスプリットモード
（ドップラー周波数制御が異なる）なのでGroup分をして制御しないと混乱するかもしれません。SatPC32ではアプリそのものを
分けています。SatPC32はサテライト用、SatPC32ISSは同一バンド用とアプリで分けています。

DATAモードについてはSatPC32でも追従できていなく（FM-Dなどは無視されFMのまま）マニュアルで無線機をデータモードに
しています。なぜかIO-117だけはSatPC32ISSでUSB-Dモード対応をしてくれました。

無線機の周波数制御はいろいろと大変そうですが（VFO-AとVFO-Bが勝手に入れ替わったり）評価の協力はさせていただきます。

githubもFreeDVアプリ（HFのSSB帯域でデジタルボイス通信ができるアプリでオープンソース開発されている）で初めてアクセス
してダウンロードだけさせていただいていました。issuesへの書き込みは今回が初めてでした。

JF1PTU, 2025-03-25:
追加でバグ関係を記載しておきます。
・トーン周波数の単位がMHzとなっています。Hzが正しいです。一般的には67.0Hzが多いです。（地上局のFM-REPは88.5Hzが多いですが、それとは異なる周波数にして妨害を防いでいます）

・周波数表示エリアのTX周波数、RX周波数の位置は逆にしてほしい。ICOMは上がRX周波数、下がTX周波数の位置になっています。現状、周波数表示（ドップラー制御含む）が有効なのはAOSからLOSまでとなっていますが、他のアプリでは常時有効で表示していますが、IC-9700の画面表示の周波数表示の明暗点滅の問題（気になる点滅で、できればやめてほしい）もあるのでAOS前制御（例えば2分）、AOS～LOS、LOS後制御（例えば2分）の間は最低でもドップラー制御継続してほしいです。（ローテータ制御時間と同期することになりますが）

・LOS後のパーク設定がN方位、仰角90度（天頂）となっています。評価に支障があるので仰角は0度（水平）としてください。現状はAUTOを外して手動変更（例えば水平に戻す）してもAUTOにするとまた天頂に戻ります。LOS後のパーク位置移動後はパーク設定は一旦解除してください。

・AOS前のローテータ制御開始時間（何分前：例えば２分前）設定ができますが、LOS後のAz制御（Elは０度まま）はそれと同じ時間だけ継続してください。継続後、パーク位置に戻し（Elは0度）終了となります。AOS、LOS前後で時間を取る理由は、アンテナをその方向に向けるまでの動作時間確保、受信ができる可能性がある、TLE制御の誤差時間などを考慮しているからです。

kei-Shima, 2025-03-26:
ご連絡ありがとうございます。issue管理の方針としてissueごとに要点を切り分けて記載いただければと思います。
今回の場合はissueと要望が混ざってしまっているので、新しいissueは新しく今のissueの番号とは分けてタイトルを問題点の内容や要望の内容とし、詳細な内容を記載いただき、要望も同じ内容なら追記いただき、新しい内容なら新しいissueで記載いただければと思います。

そうしないとお互いの取りこぼしや理解の不一致につながりやすくなってしまいます。

rsp-daigo, 2026-02-01:
本Issueは一旦クローズさせていただきます。
ご指摘の内容の内、未対応のものについては以下Issueを作成しましたので、そちらで対応を進めたいと思います。

①[無線機設定画面における接続テストの正確性の改善](https://github.com/RymansatSatelliteTracker/RST/issues/168)
②[ローテータ設定画面における接続テストの正確性の改善](https://github.com/RymansatSatelliteTracker/RST/issues/169)
③[ローテータのフリップモード対応](https://github.com/RymansatSatelliteTracker/RST/issues/170)
④[無線機制御の受信固定、送信固定に対応する](https://github.com/RymansatSatelliteTracker/RST/issues/171)

closed 2026-02-01 by rsp-daigo