RymansatSatelliteTracker/RST #8: 使用させてもらいました。

yutaka5913, 2025-03-24:
リーマンサットの皆さんFBな追尾ソフト有難うございます。

インストールと初期設定。
SatPC32と比べれば、圧倒的に設定が簡単です。マニュアル無しで簡単に設定できました。とても使い易いです。最初から予定されていた通りの出来上がりですね。

設定はほぼわかりますが、衛星情報の設定方法がわかりません。送信周波数とか受信周波数を少し変化させたいです。

とりあえず、ローテーターコントロールを使わないで（ローテーターは我が家はS.A.Tでコントロール）SO124でQSOできました。
周波数がおかしいです。RSTで表示されている周波数とIC-9700で設定される周波数が違います。ドップラーの関係でしょうか？
それとアップリンクとダウンリンクの周波数が数KHzずれています。

その後何回か使っていたら、SO-124でRSTの表示は　145.927/436.876　なのに　IC-9700の表示が　144.06/436.999で動かなくなってしまいました。IC-9700の電源を切って立ち上げ直しても、RSTでサテライトモードに入るとこの周波数が設定されて動きません。
しかたないので、再インストールをしようと思っています。

こういうトラブルはありますが、ソフトウエアは素晴らしく。英語で公開したら全世界に広がりそうな感じがします。

いろいろと言ってますが、追加機能のお願いです。

①　グループ内の一つのサテライトが行ってしまった後に　NEXT Satellite　というボタンを押すとグループ内の次に来る衛星に切り替わってほしい。次に来る衛星が気にいらなければ、もう一回そのボタンを押すとその次に来る衛星に切り替わってほしい。　これはS.A.Tに付いている機能でとても便利に使っています。NEXT Satellite　ボタン追加です。
②　友人がBTRotatorに繋いで実験していますが、衛星がLOSで見えなくなった時にAZ=0　EL=0　に強制的に戻りますが、これは戻さないで見えなくなった方向に向いたままの方が良いと思います。水平線に沈んだ後もずっと追っかけていたいものです。

yutaka5913, 2025-03-24:
名前忘れました。　JA1COU　村田です。

kei-Shima, 2025-03-25:
ご連絡ありがとうございます。周波数設定の不具合は複数報告いただきましたので確認させていただきます。

衛星情報の設定というのは周波数設定でしょうか？　「衛星設定」画面で衛星の名前をダブルクリックすると設定画面が表示できます。
[image]

無線機周波数が動かなくなる問題は初なので確認させていただきます。もしかすると追加情報をお願いするかもしれません。

ご要望の「Next Satellite」は面白い機能かと思います。画面サイズ等もろもろの確認がいりますのでいったん要望として検討させていただきます。

JF1PTU, 2025-03-25:
S.A.TのNext Satelliteと同様の機能として、SatPC32ではCountdownという子画面でアプリに登録している衛星（A～X：最大24個登録可）すべてのAOSまでの時間とMEL（最大仰角）を時系列に一覧表示します。AOSになると青背景色に変わりLOS時間までの残り時間を表示します。LOS後は最上段から消え、次のAOSまでの時間表示となり順番が下のほうに表示されます。MELの閾値設定もできます。（MELが閾値以下の衛星は表示しない設定が可能）残念ながら衛星切り替えは画面を見ながらマニュアル設定です。あれば便利な機能です。一覧表示のオンオフもできます。要望とします。de JF1PTU大澤

kei-Shima, 2025-05-11:
> その後何回か使っていたら、SO-124でRSTの表示は　145.927/436.876　なのに　IC-9700の表示が　144.06/436.999で動かなくなってしまいました。IC-9700の電源を切って立ち上げ直しても、RSTでサテライトモードに入るとこの周波数が設定されて動きません。

一旦、この現象は発現していないのでCloseさせて頂きます。
また、現象が発現したらissueを入れて頂ければと思います。

closed 2025-05-11 by kei-Shima