RymansatSatelliteTracker/RST #9: 再起動しなくなりました。

JF1PTU, 2025-03-25:
その後、RSTアプリを再起動させましたが起動しなくなりました。仕方がないので
アンインストールしようとしましたが、PCはアプリが起動していると判断していて
アンインストールできない状態です。（マニュアルでの強制消去もできません）
現状、この状態でRSTアプリの評価が出来なくなりました。

タスクマネージャで見るとアプリは立ち上がっていなく、バックグランドプロセスで
rstロゴマークが3個の倍数で6個、9個と起動させる都度に増えていくだけで初期画面
の地図が現われません。
これ以上は、私のPCスキルでは、なぜ起動してくれないのか分かりません。
PCはWin10（Ver22H2）で品質累積更新プログラム（KB5053606）

issueリストに書き込んでおきます。

無線機の周波数制御に関しては、無線機側のダイヤル周波数情報はRST画面の右上周波数表示部分に
送られるようです。RSTアプリ側のドップラー周波数制御情報がうまく無線機側に送られていない
ようです。データパケットを連続的に送ると無線機側で処理しきれない？SatPC32ではCAT Delay
設定（送信待ち時間）があり20msを設定しています。（通信速度で変わるようですが）

SatPC32のカウントダウンリスト(次の衛星が何分後に来るか、maxEL（最大仰角）の表示）の機能は
あれば便利です。

JF1PTU, 2025-03-25:
CTRL+ALT+DELでRSTアプリを強制終了しRSRアプリのアンインストール後、再度、インストールしRSTアプリの起動を試みましたがやはり初期の地図画面は出ませんでした。他のアプリは正しく起動しています。

JF1PTU, 2025-03-25:
RSTアプリが起動していない時のタスクマネージャのバックグランドでのrst.exeが3個、実行中となっています。メモリ容量は
夫々39MB/9MB/5.5MBです。

[image]

JF1PTU, 2025-03-25:
DTPCを用いてのRSTアプリの評価はとりあえず保留し、NotePCで評価を再開します。

kei-Shima, 2025-03-25:
ご連絡ありがとうございます。
取り急ぎ原因究明に向けて確認作業進めさせていただきます。
ノートPCでも同じ不具合が出るようでしたらお手数ですがお知らせください。

kei-Shima, 2025-03-25:
取り急ぎ、アンインストールしていただいた後で、以下のフォルダの設定ファイルを消していただくとソフトは初期に戻りますが地図は表示されるかなと思います。お試しいただけますか？

削除するフォルダ[rst]
C:\Users\”ユーザー名”\AppData\Roaming\rst
”ユーザー名”は登録されているユーザー名に置き換えて入力してください。

[image]

JF1PTU, 2025-03-29:
想定原因を＃14に記載しました。（TLE内に登録済みの衛星がなくなると起きるようです）

ja1cpa, 2025-04-02:
Win11でrstを削除する方法（当局の場合）

1. コントロールパネルでrstをアンインストールする。
これだけでは、全てのrstファイルは削除されない模様。

2. RSTを停止させる
「Ctrl」+「Alt」+「Delete」同時押し
「タスクマネジャー」クリック
「プロセス」一覧表が出る
一覧表左側の
「R」クリック、右上の「◯タスクを終了する」クリック
「R」が消える
「R」クリック、右上の「◯タスクを終了する」クリック
「R」が消える
これを行の一番下まで全て行う。

3. 隠しフォルダの表示（AppDataが出ていなかった）
エクスポーラを表示する
一番上の中ほどの「表示」クリック
一番下の「表示」にマウスを！
右に展開するので一番下に「隠しファイル」が出る
マウスを移動しクリックしチェックを入れる

4. rst消去（2か所に有った。rstをクリック、消去アイコンをクリック）
C:\User\tv3-\AppData\Local,Roaming　 (LocalとRamingが同時に出た) 
C:\User\tv3-\AppData\Roaming\Pograms\rst　（これでrstファイルが全て消える）
C:\User\tv3-\AppData\Local\Pograms\rst   (これでrstファイルが全て消える）
　　　　↑tv3-はそのPCによって違う

当局は、C:\rst\    C:\CALSAT32\    のようにC:直下に設定して使っています。

kei-Shima, 2025-05-11:
TLEのファイルによってはTLE以外の文字が含まれている部分ができずに起動できていませんでした。
現在は処理できているのでCloseとさせていただきます。

closed 2025-05-11 by kei-Shima