RymansatSatelliteTracker/RST #11: IC-9700の送/受信周波数を逆に設定

jh4xsy, 2025-03-26:
SO-124をトラッキング中のRSTスクショ。437MHz帯はRX側。
[image]

IC-9700では、TX側に437MHz帯が設定される。
[image]

JF1PTU, 2025-03-26:
SO-124(HADES-R)のアップリンク周波数は145.925MHz、ダウンリンク周波数は436.888（436.885？）MHzです。トラッキング中とのことですが全くズレており、また、送受信周波数がひっくり返っています。おそらくCI-Vコマンドエラー（周波数制御エラー）が発生していると思われます。

IC-9700との接続を外して、RSTアプリのみでトラッキングさせると、正しくドップラー制御がかかった周波数表示になります。従って、周波数制御のCI-Vコマンド送信・受信エラーの改善が必要と思われます。

ja1cpa, 2025-03-27:
衛星の位置と軌跡は正しく動作しています。AO-10も軌跡（楕円軌道）を表示できました。非常に素晴らしくできています。
まだ途中ですが、
1. 衛星設定で周波数を設定してもトップ画面の右上欄にすぐには反映しません。（Auto/Satelliteクリック）
2. 衛星の周波数設定でビーコン周波数の設定欄がありません。最近の衛星はビーコン専用周波数は無いのが多いですが。世界的には？
3. 衛星の周波数設定で設定によってはエラーになって設定できません。
4. トーン周波数がMHzになっています。（Hzにして！）
5. IC-9700をONするとトップ画面右上の周波数がIC-9700の周波数に設定されてダイヤルを動かすと変化します。
6. 最初はRSトラッカーからIC-9700に設定してダイヤルを動かしたらそれをRSトラッカーへ！（S.A.Tトラッカーは動かした周波数からドップラー変化する）
7. トップ画面を含むどこかでTXとRXが入れ替わっているようで頭が混信しています。
8. rsp antenna IOにローテーターを繋がずにテストしましたが動作しませんでした。DIN8Pコード入手中。
de ja1cpa

jh4xsy, 2025-03-27:
@JF1PTUさん, @ja1cpa さんのIC-9700では送受信周波数は正常＝ひっくり返ってないのですか？

jh4xsy, 2025-03-29:
下記条件だと、送受信のひっくり返りは発生しません。
- src/common/Constant.tsで、MAIN_BANDとSUB_BANDの値を入れ替え
- RSTの送信周波数帯=IC-9700のサテライトモードでの送信周波数帯

IC-9700のサテライトモードではTXはSUBになってます。検証に使ったスクリプトの実行結果を貼っておきます。
```
$ python3 test2.py 
IC-9700 CI-V制御TEST開始
ポート: /dev/ttyUSB0, ボーレート: 115200
シリアルポートを開きました。

ステップ1: サテライトモードをONにします...
送信: FE FE A2 E0 16 5A 01 FD
受信: FE FE E0 A2 FB FD
-> コマンド成功 (FB)

ステップ2: サブVFO (TX側) を選択します...
送信: FE FE A2 E0 07 D1 FD
受信: FE FE E0 A2 FB FD
-> コマンド成功 (FB)

ステップ3: 送信周波数を 145.925 MHz に設定します...
  生成されたBCDデータ: 00 50 92 45 01
送信: FE FE A2 E0 05 00 50 92 45 01 FD
受信: FE FE E0 A2 FB FD
-> コマンド成功 (FB)
送信周波数の設定に成功しました。

処理完了。
シリアルポートを閉じました。
TEST終了。
```

JF1PTU, 2025-03-29:
各コマンドの送信間隔はどのように設定されていますか？無線機からの応答コマンド（FB:OK、FA:NG）を受けてから次の送信コマンドを送るようなシーケンスであれば問題はないように思いますが？

SatPC32アプリでは、送信間隔（CAT delay）を無線機と通信速度により以下のように変えるようにアドバスしています。

bb. Enter a proper CAT delay value.
Try the following delay values:

Yaesu FT-736R: 20,

FT-847. 30 at 4800 or 9600 baud, 20 at higher baud rates.

FT-817/857/897:  110 at 38400 baud.

Kenwood TS-790: 70

TS-2000: 20 at 57600 Bd.  

 Icom IC-820H, IC-821H, IC-910H, IC-9100: 40 – 90 at 9600, 20 – 70 at 19200 baud, 140 at 1200 baud.

IC-9700で115200 baudの条件は判りませんが20に設定しています。（単位はmS？）

CI-Vコマンドは、元々はバス方式なので（送受信同じ信号ライン）、半二重通信ですが、双方が同時送信した時はコマンドがぶつかりエラーが起きるのではと思います。

いずれにしても無線機を切り離して動作させた場合、ドップラー周波数制御は、Jモード（145送信/435受信）でもBモード（435送信/145受信）、デジピートモード（145送信/145受信、435送信/435受信　ただしサテライトモード設定）でもほぼ正確にRST画面の右上に周波数表示されています。（SO-50など一部の衛星ではRX周波数が1kHz台でズレることがありますが）

送受信周波数が期待値ではないので実際の送信は行っていません。まずは1Hzまでの周波数表示を増やす（3桁増やす）ことが先決かもしれません。

jh4xsy, 2025-04-01:
送信周波数の設定に関して、rigctlではIC-9700が"サテライトモードで↑435MHz/↓145MHz"な初期値でも、手動で↑145.925MHzに設定できるので、詳細ログを取ったら、追加(裏)コマンドを発行してた。なる程です。
　
```
$ rigctl -r /dev/ic9700a -s 115200 -m 3081 -t 4533 -vvvvv
...
0000    fe fe a2 e0 05 00 50 92 45 01 fd                    ......P.E.. ★ １回目    
0000    fe fe e0 a2 fa fd                                   ......          
icom_transaction: failed: rig_get_freq(2230): freqMainA=145850000, modeMainA=FM, widthMainA=12000
rig_get_freq(2230): freqSubA=435400000, modeSubA=, widthSubA=0
0000    fe fe a2 e0 05 00 50 92 45 01 fd                    ......P.E.. ★ ２回目    
0000    fe fe e0 a2 fa fd                                   ......          
rig_get_freq: Sub cache hit age=54ms, freq=435400000, use_cached_freq=0
0000    fe fe a2 e0 07 b0 fd                                .......     ■ MAIN/SUBの入替！コマンドを投下    
0000    fe e0 a2 fb fd                                      .....           
0000    fe fe a2 e0 05 00 50 92 45 01 fd                    ......P.E.. ☆ ３回目    
0000    fe fe e0 a2 fb fd                                   ......          
rig_get_freq: cache miss age=10000ms, cached_vfo=Sub, asked_vfo=Sub, use_cached_freq=0
...

kei-Shima, 2025-04-21:
各位
こちらの修正がマージされております。ビルド可能な方はお試しいただければと思います。

[https://github.com/RymansatSatelliteTracker/RST/pull/35](https://github.com/RymansatSatelliteTracker/RST/pull/35)

jh4xsy, 2025-04-21:
ビルドして試したところ、IC-9700のサテライトモードのTX/RX周波数の組合せを変えても、RSTの送/受信周波数は、IC-9700 サテライトモードのTX/RXに正しく設定されました。TNXです。

closed 2025-04-21 by jh4xsy