RymansatSatelliteTracker/RST #14: TLE読み込みの更新失敗の後、アプリを終了させ、再起動させると再起動しませんでした。

JF1PTU, 2025-03-29:
評価を初めて、RSTアプリが再起動しない現象が数回、発生しました。発生条件が解りました。おそらく、TLE読み込みのURLを変えてOKボタン押すと更新が失敗の表記が出たあと、アプリを終了させ、再起動させると再起動しなくなりました。原因は、当初のURLで読み込んだTLE情報で衛星名を選択後、URLを変えてアプリを再起動した時に、その衛星名が存在しないとおかしくなる（再起動しなくなる）のではと推察しています。
CALSAT32でも、登録済みの衛星が消滅した時にTLE情報にその衛星がないとエラー情報が出ます。場合によっては起動しなくなったように思います。

今まで使っていたアプリ（CALSAT32、SatPC32、SatPC32ISS）ではamsat系のアマチュア衛星のURLでTLEを読み込んでいましたが。RSTアプリでは
celestrak系の各種衛星のURLを読み込んでいるようで、衛星名もISSでなくISS(ZARYA)、ISS(NAUKA)が出てきます。

異なるいくつかのURLで同時にTLE情報を読み込んだ時に同じNORAD IDが出てきた時は、どちらかの衛星名でTLEが登録されダブらないようにしているのでしょうか？

★再起動しなくなった時の対処方法★
1.タスクマネージャを起動し、バックグランドプロセスタスクに残っているrstのアイコン（3個以上残る）を全て終了させる
2.rst.exeが登録されているフォルダ（PC>Wiondows>Users>USER>AppDAta>Local>Programs>rst)内のUninstall rst.exeを起動させ、フォルダ内のファイルを全てアンインストールさせる。（1項の手順をやっておかないと全てアンインストールできないので注意）
3.rstフォルダ内のデータを全て消去する。
4.RSTアプリを再インストールし設定をやりなおす。（衛星名と周波数の登録がめんどうですが）

おかしくなる前に、app_config.jsonが登録されているフォルダ（PC>Wiondows>Users>USER>AppDAta>Roming>rst）内の同ファイルのバックアップを取っておき、再インストール後に、同ファイルをバックアップしていたファイルと置き換えれば、再設定は不要になります。
（おかしくなった時に同ファイルのバックアップを取っても、おかしくなった時の状況が保持されているので、置き換えてもダメでした。おかしくなる前にバックアップを取ることをお勧めします。）

kei-Shima, 2025-04-01:
ご連絡ありがとうございます。
こちらでも確認中ですが、再起動しない部分を詳しく教えていただけますか？

以下のような感じでしょうか？
アイコンをダブルクリック　→　ウインドウは立ち上がるが地図が出ない
それとも、ダブルクリックしてもプロセスだけ立ち上がって画面に何も出ない
等でしょうか？
もし、何か画面に出ているようならスクリーンショットも頂ければ参考になります。

JF1PTU, 2025-04-01:
発生条件が解りました。

初めてインストール後、TLE読み込み画面の8つあるURLの最後のURLを消去して、普段使っているURLを貼付けクリックマークを入れてOKボタンを押すと、しばらくしてエラー表示（更新失敗）が出て（これが発生条件のようです。すぐに消えるのでスクリーンショットできていません）、アプリを終了させ、再起動しても画面は立ち上がらないようです。衛星登録は初期のままで、新たな衛星登録はしていません。

RSTアイコンをダブルクリックする都度、タスクマネージャのバックグランドプロセスのrstアイコンが増えるだけです。アプリ側のプロセスにはタスクマネージャのみです。

ちなみに追加した（書き換えた）URLは、https://www.amsat.org/tle/current/daily-bulletin.txt（毎日TLEが更新される）です。

JF1PTU, 2025-04-01:
スクリーンショットが取れましたので添付します。

[image]

ja1cpa, 2025-04-02:
ja1cpaです。私も同じことをやっていました。hi

kei-Shima, 2025-04-18:
以下のURLのTLEの頭にメッセージが入っており、それをTLEとして読み込もうとしてエラーとなっているようです。
https://www.amsat.org/tle/daily-bulletin.txt
ソースコードの修正は終わりましたがバイナリのリリースはまだ行っていないので次のリリースv0.1.2までお待ちいただくか、ソースからビルドしていただければと思います。

jh4xsy, 2025-04-18:
Linux(Leap 15.6)にて Commits on Apr 18, 2025 に更新、$HOME/.config/rstを削除した後、TLEにdaily-bulletin.txtを追加しましたが、読み込みエラーなく、RSTも再起動できました。TNXです。

kei-Shima, 2025-05-11:
＃９と同じ問題でしたのでCloseさせて頂きます。

closed 2025-05-11 by kei-Shima