RymansatSatelliteTracker/RST #27: ローテータ画面の機種設定と動作設定のタブ表示がでない（Ver0.1.1）

JF1PTU, 2025-04-07:
##概要
issueの概要を記載してください。issueが複数の場合はissueごとに分けてください。

クリーンな状態で（前Verの2箇所のフォルダ内を全て削除した状態）Ver0.1.1をインストールしたが、新しく追加されたタブが表示されない。
ただし、Ver0.1.1を上書きインストールした場合は（Win10のDTPC）、機種設定/動作設定タブは表示される。

##使用環境
使用環境を記載してください。最低限OSは記載いただければと思います。
例：WIndows11、MAC、Ubuntsu

Win11（NotePC）　　　Winのバージョンは関係ないと思いますが上書きでのVerUPでは問題なしでした。　　

#RSTのバージョン
例：0.1

Ver 0.1.1

##issueの画面
issueが発生した画面のスクリーンショットを貼り付けてください。

ローテータ画面のローテータの文字の下の、「機種設定」、「動作設定」のタブ枠がない状態
表示内容は機種設定の画面内容

##行った操作
どのような操作を行った結果、issueが発生したか記載してください。

Win11のNotePCにクリーンな状態（前のVerを全て消去）でインストール後、ローテータアイコンをクリックし、。ローテータ設定画面を表示させた時に発生。

JF1PTU, 2025-04-08:
RSTアプリをアンインストールし、かつ、roaming>rstフォルダ内も全て消去し、再インストールでタブが表示するようになりました。前回のインストールで何か問題があったのかもしれません。

rsp-daigo, 2026-01-04:
@JF1PTU 
本件、現状は問題なしと思われますのでクローズします。
問題アリの場合は別途Issueを上げていただければと思います！

closed 2026-01-04 by rsp-daigo