RymansatSatelliteTracker/RST #28: ローテータ機種設定画面のテストモード時のAz動作用の左矢印ボタンが動作しない（ｖ0.1.1）

JF1PTU, 2025-04-07:
##概要
issueの概要を記載してください。issueが複数の場合はissueごとに分けてください。

ローテータ機種設定画面のテストモード時、Az動作用の左矢印ボタンを押しても左側に（360度~180度S側）動かない。ただし、Azの数字入力枠の0度を360度に書き換えて左矢印ボタンを押した場合は360度から180度S側に向かって動作はする。（ST-2もBT-AZ/ELも同じ動作はする）

Azの数字入力枠の数字はマイナス数字入力しても0度となる。（両機種設定とも）
ただし、BT-AZ/ELはAz入力枠の上の応答表示（たぶん応答表示数字）は、0度から-450度、あるいは0度から450度表示するようです。Elは-10度~190度の入力に対して-10度~180度~-170度の応答表示なるようです。

ST-2はマイナス表示はしない仕様のようです。仰角を-10度あるいは190度としても物理的には0度~180度に制限がかかるようになっているようです。

##使用環境
使用環境を記載してください。最低限OSは記載いただければと思います。
例：WIndows11、MAC、Ubuntsu

Win10、Win11

#RSTのバージョン
例：0.1

ｖ0.1.1

##issueの画面
issueが発生した画面のスクリーンショットを貼り付けてください。

##行った操作
どのような操作を行った結果、issueが発生したか記載してください。

ローテータ機種設定画面で接続テストボタンを押して、接続成功後、左矢印ボタンを押しても動作しない。
（もしかして正常仕様？？）

rsp-daigo, 2025-05-24:
@JF1PTU さん
本件確認させてください。

> ローテータ機種設定画面のテストモード時、Az動作用の左矢印ボタンを押しても左側に（360度~180度S側）動かない。ただし、Azの数字入力枠の0度を360度に書き換えて左矢印ボタンを押した場合は360度から180度S側に向かって動作はする。（ST-2もBT-AZ/ELも同じ動作はする）

こちらですが、「0°（北）からマイナスとなる西方向へ動かない」という理解で合ってますでしょうか。
現状（v0.1.2）ですと、テストモードのAz入力値は「0～450」までとしているため、以下の挙動となります。
- 入力値が0°の状態：　「←」を押しても入力値は0°のままとなる。
- 入力値が360°の状態：　「←」を押すと359°になる。アンテナも西方向に変化する。
本件は、これに起因したご指摘となりますでしょうか。

JF1PTU, 2025-05-25:
ご理解の通りです。Azローテータは起点位置の関係で2種類のローテータがあります。

その1：起点位置がS（南）のものでKENPRO製KR-400（古い機種です。中古しかありません）
アナログメータ表示　S(180度)⇔N(0度/360度)⇔S(180度)　従ってパークポジションをN(0度)とするとアナログメータ針はN=0度/360度時は真ん中になるため視覚的には左矢印ボタン押下時は左側に動作して欲しいとなります。

その2：起点位置がN（北）のものでYAESU製G-5500DC（現在入手可能）
　アナログメータ表示　N(0度)⇔S(180度)⇔N(360度/0度)⇔E(450度/90度）　従ってパークポジションをN(0度)とするとアナログメータ針は左端となるので視覚的にも左矢印ボタンを押しても、動作しなくても問題ありません。

現状の仕様は「その2」に合っていることになりますが、「その1」にも対応させるには、アプリの初期設定でAzの起点位置（Turning Potion）がNかSかの選択、Azが0度~360度か0度~450度かの選択ができるようになっています。またEl（仰角）も０度~90度か0度~180度かの選択ができるようになっています。

Fox Delta社のST-2は、その初期設定をST-2側でやるようになっています。従ってコマンドでAz＝350度をST-2に送ればアナログメータは左側に動きます。

現状のRSTのローテータ設定画面の矢印ボタンは、あくまでテスト用と割り切れば、問題ないかと思います。(最近のものは起点位置設定がNのものがほとんどなので）メイン画面のレーダチャート画面のマニュアル設定（AUTOをオフ）にしてマウスでクリックすれば、その位置（方角、仰角）に移動するので、動作確認はできます。

rsp-daigo, 2025-05-25:
@JF1PTU さん
ありがとうございます。
> その1：起点位置がS（南）のものでKENPRO製KR-400（古い機種です。中古しかありません）
理解しました。検討します。。