RymansatSatelliteTracker/RST #63: リニア衛星のトラッキング

yutaka5913, 2025-05-01:
リニア衛星のAUTO時のドップラー補正ですが、受信機からのダイアル情報を取り出してそこからがスタートになります。

通常リニア衛星でQSOする場合、自分がCQを出して呼ばれてQSOする場合とダイアルをぐるぐる回して、CQを出している局を見つけてQSOする場合があります。

RSTでは、次の様に動いてもらいたいと思います。

まずトランシーバーの受信周波数を読み出す。

次に、受信周波数での衛星とのドップラー補正周波数（差）を計算します。

RIGの受信周波数にドップラー補正を加えます。これが、衛星が送り出している送信周波数です。

送信周波数がわかれば、トランスポンダーの周波数変換でトランスポンダーが受信している周波数がわかります。

今度はトランスポンダーが受信している周波数の地上局とのドップラー補正値を計算します。

トラポンの受信周波数にこのドップラー補正値を加えます。

その周波数を、トランシーバーの送信周波数にセットします。

この動作を一秒に一回やれば良いのではないでしょうか？

リニア衛星の場合は、受信機のダイアルをぐるぐる回してQSOするので、FM一チャンネルの衛星のドップラー補正のやり方とは、違います。

そうしないと使い物になりません。SatPC32はそういう動きをします。

FM衛星の衛星固定の周波数補正のやり方とは違います。

FM衛星は基準となる周波数は衛星の送受信機の周波数。
リニア衛星は、運用局の受信周波数が基準となるということです。

私は、トラポンとかトラッカーを設計したことはないのですが、地上局の運用方法から推測してのお話です。 

JA1COU/村田

kei-Shima, 2025-05-10:
ほかのソフトと同じような挙動で操作可能か検討させて頂きます。

rsp-daigo, 2025-11-07:
#116 にて対応しました。